SpiNNakerManchester/JavaSpiNNaker #429: Enhancement: Spalloc Server could Reload FPGA Bit files on error

rowleya, 2022-03-03:
A non-critical future enhancement request.  Every now and again, the flash memory loses the FPGA configuration data and it has to be reflashed.  When this happens, the FPGA ID flag comes back as 3 (which is why spalloc checks these flags when the machines are being allocated).  Currently if the read fails, the allocation is given up, but an alternative would be to automatically perform the steps that are currently done manually to reload the bit files.

Information:
 - The current bitfiles (*.bit)and bmpc script (bt_load) are here: https://github.com/SpiNNakerManchester/SpiNNaker_hardware_tests/tree/master/firmware
 - The bmpc implementation that loads these is here: https://github.com/SpiNNakerManchester/spinnaker_tools/blob/master/tools/bmpc

dkfellows, 2022-03-11:
I'm now sure that `comms` _doesn't_ support all of those. I'd done enough for ybug and spalloc and the main toolchain, but not for bmpc. That has quite a few extra operations; this will take some time…

dkfellows, 2022-03-31:
That PR should do it. I'm just uncertain how to test it without causing big problems if things go wrong. I don't trust myself to get it right first time... 😉 

Also not sure if we should reload the FPGA the first time it fails to boot right, or if we should try a simple reboot first.

rowleya, 2022-04-01:
I think that trying this on the 12 board system should be fine; I don't think it can mess up the BMP (at worst you should get an error message), so there will always be a way to recover.  The FPGAs might have incorrect bitfiles, but they should be replacable using the BMP anyway.  @lplana may correct me though!

I think if the FPGA id is read wrong, it won't fix itself, so as soon as this is detected it should be possible to overwrite.

I am unsure how this can be tested since the failure itself is rare.  A possible idea is to make it load every time to check it I guess...

lplana, 2022-04-01:
Just checking that I'm on the right track: there are two different scenarios and each requires a different action.

- An FPGA has ID in [0, 2] but it is the wrong ID. This means that the FPGA got the wrong bitfile during power up. A power cycle should solve the problem. Spalloc currently detects this problem and power-cycles the board.
- An FPGA has ID = 3. This means that the wrong bitfile (or no bitfile) is stored in flash memory.  As @rowleya indicates, this scenario will not fix itself as it requires a flash memory write.

The failure can be caused artificially by deleting the bitfiles in the flash memory or by replacing them with different ones. The `bt_load` BMP script is used to load the correct bitfiles. A similar script`'bt_load_fault` could load `default.bit` into slots 0-3 and that should trigger the fault
.

rowleya, 2022-04-01:
OK, thanks for the clarification; I had forgotten about the ability to reboot the board if the ID is incorrect.

If we test by forcing it to load the bitfiles every time, and something goes wrong, I think it is unlikely to break anything in a serious way i.e. in which we couldn't fix by just using BMPC.  Would you agree that this is the case?

Of course we don't want to do this on SpiNNaker 1Million, but maybe on the 12 board system just to check that it could work if needed.

lplana, 2022-04-01:
> If we test by forcing it to load the bitfiles every time, and something goes wrong, I think it is unlikely to break anything in a serious way i.e. in which we couldn't fix by just using BMPC. Would you agree that this is the case?

This should be safe. Just be aware that the flash memory has a limited number of write cycles, probably around 100K ;-).

The bitfiles are stored in the board flash but the xboot configuration is stored in the BMP internal flash. The xboot command updates this memory. Repeating this many times should not be a problem, but we can always re-flash the BMP through JTAG if something goes wrong.

Loading a bitfile is somewhat slow. It can take a couple of minutes. The process is verified using a checksum and it can fail occasionally, so the result code should be checked and the process retried on failure.

dkfellows, 2022-04-01:
I don't mind things being slow (as thus is for the last-ditch case where total failure is the other option) but that it is only necessary when the ID is 3 is critical info. Thank you very much!

The structure of the server means that this will only negatively impact allocations on the frame (and things allocated upon it). Other frames should proceed just fine in parallel, unless there are a very large number of reloads wanted at once. But that's a the-catastrophe-has-already-happened case...

dkfellows, 2022-04-25:
The implementation is there but not enabled (nor am I sure this is safe to enable yet; the reloading isn't tested at all) so this issue remains open. I needed the improvements to the BMP coordinates.

dkfellows, 2022-08-04:
Some testing now done, though not yet live. If the comms layer works (which it should, assuming I got the key constants correct; it's copied from other bits that do work and follows standard patterns) then the reloading will work. In any case, if it doesn't it's now a bug.

PR #440 (2022-04-25): Reload FPGA definitions
fixes #429 

Note that this is _not_ a replacement for `bmpc`. That has things it does that this code doesn't support.
closed 2022-08-04 by dkfellows