SpiNNakerManchester/JavaSpiNNaker #441: Proxy

dkfellows, 2022-03-30:
We should be able to ask spalloc server to put up a proxy (HTTP-based for ease of getting through firewalls?) to our job so that we can interact with boards on the big machine from outside the university despite the boards being on internal IP addresses. This would allow people to, say, use robots with the big machine, provided they can handle the latency.

dkfellows, 2022-03-30:
Notes
-
Java: https://spring.io/guides/gs/messaging-stomp-websocket/
Python: https://pypi.org/project/stomp.py/

rowleya, 2022-03-30:
Note the data that would flow over the websocket is likely to be binary data rather than text.  This would likely need to be encoded with STOMP.  Alternatives are to use binary handling e.g.:
https://developpaper.com/spring-boot-websocket-custom-protocol-development/
https://websockets.readthedocs.io/en/stable/

Note also that regular http/REST is likely fine for most of the interactions e.g. boot machine, read version, read/write memory etc. since most of those are based on sending messages and getting responses.  The only time a websocket is likely needed at all would be for live output where data is then streamed from the machine; even then long polling *might* work (though it might be a bit too slow for interactive use cases).

dkfellows, 2022-04-19:
Turns out that [websocket-client](https://pypi.org/project/websocket-client/) is a better Python package for our purposes. Specifically, we now use that and it works.

The client itself is in SpiNNMan because it tangles with proxy transceivers. Putting the basic comms in there feels like the right place anyway.

dkfellows, 2022-04-26:
Oh EIEIO, how I dislike thee!

dkfellows, 2022-04-26:
The Live Packet Gatherer wants to open an _unbound_ `EIEIOConnection`, use it to contact SCAMP on each of the boards to set up a tag to talk to it, and then have EIEIO messages sent from SpiNNaker to it that way. That's horrible to handle with the current proxying mechanisms, as it involves unbound sockets, send-to operations, and protocol punning. 🙁 

I'm working on it, but it will take some time.

rowleya, 2022-04-27:
This sounds like the firewall poking mechanism.  The idea is that the EIEIOConnection will be bound to a local port and then it wants to talk to the SpiNNaker machine on the SCAMP port, which ensures that packets are returned to the same connection.  This will poke through firewalls and NAT routers.

Note that the EIEIOConnection should be bound to a local IP address and port in any case;  I am unsure I am seeing where the unbound bit comes from...

dkfellows, 2022-05-15:
I've been gradually looking through this. There are two parts to the LEC, one of which is the notification protocol stuff (because it inherits from `DatabaseConnection` &mdash; that is a *terrible* class name as it is *misleading* as to what is going on, and calling it a notification protocol participant would be better, but fixing it would be a pain as it is more user-exposed than most parts) and the other of which is basically an unbound `EIEIOConnection` that the relevant `IPTag`s are programmed to target. (There is also a separate phase of sending a message to make the firewall not interfere. That's a bit that isn't needed with the proxied connections; they're already through.) The old style buffer manager does something similar, except without the bonus confusion of the notification protocol stuff in the same class hierarchy. 

Which is all messy and complicated. 

I can't remember which part of the code uses bound `EIEIOConnection`s.

rowleya, 2022-05-16:
Note that the firewall poking may not be needed at all with the IP tag re-programming, since the firewall poking just needs a message to be sent from the socket doing the receiving.  I think the code for firewall poking was added first, and then never removed with the change to the IP tag reprogramming (it doesn't break anything so it was probably not noticed).

PR #444 (2022-04-20): Proxy
An integrated system for setting up proxies to SpiNNaker boards. Uses web sockets (will be the `wss:` basic protocol because we're using HTTPS to establish) to provide connection establishment and basic message framing.

fixes #441

Supported _binary_ protocol (within overall WebSocket framing)

<table border>
<tr>
<th>Name</th>
<th>Request Layout (words)</th>
<th>Response Layout (words)</th>
<th>Meaning</th>
</tr>
<tr>
<td>Open Conn.</td>
<td><table border>
<tr>
<td><code>0</code>
<td>CorID</td>
<td>X
<td>Y
<td>UDP<br>Port
</tr>
</table>
(5 words)
</td>
<td><table border>
<tr>
<td><code [... cut]
closed 2022-08-04 by dkfellows