SpiNNakerManchester/JavaSpiNNaker #453: Timeout doesn't get triggered on download

rowleya, 2022-04-11:
This was run with the reduce_overhead branch but I don't think the code differs in any important way here from master.  The code freezes on download.  The jspin log is apparently too big to attach, but a jstack has been output and attached.

[jstack.txt](https://github.com/SpiNNakerManchester/JavaSpiNNaker/files/8462808/jstack.txt)

dkfellows, 2022-04-26:
The one place where things seem to be blocked is at this thread in the stack trace (nothing else suspicious is present):
```
"pool-2-thread-7" #62 prio=5 os_prio=0 cpu=31549.07ms elapsed=53817.93s allocated=10551M defined_classes=1 tid=0x00007f71e8913800 nid=0x1124f runnable  [0x00007f703fcfb000]
   java.lang.Thread.State: RUNNABLE
	at sun.nio.ch.EPoll.wait(java.base@11.0.11/Native Method)
	at sun.nio.ch.EPollSelectorImpl.doSelect(java.base@11.0.11/EPollSelectorImpl.java:120)
	at sun.nio.ch.SelectorImpl.lockAndDoSelect(java.base@11.0.11/SelectorImpl.java:124)
	- locked <0x000000008f27b2c0> (a sun.nio.ch.Util$2)
	- locked <0x000000008f27a4d8> (a sun.nio.ch.EPollSelectorImpl)
	at sun.nio.ch.SelectorImpl.select(java.base@11.0.11/SelectorImpl.java:136)
	at uk.ac.manchester.spinnaker.connections.UDPConnection.readyToReceive(UDPConnection.java:642)
	at uk.ac.manchester.spinnaker.connections.UDPConnection.isReadyToReceive(UDPConnection.java:611)
	at uk.ac.manchester.spinnaker.connections.UDPConnection.doReceive(UDPConnection.java:304)
	at uk.ac.manchester.spinnaker.connections.UDPConnection.receive(UDPConnection.java:288)
	at uk.ac.manchester.spinnaker.front_end.download.GatherDownloadConnection.getNextPacket(GatherDownloadConnection.java:138)
	at uk.ac.manchester.spinnaker.front_end.download.DataGatherer$Downloader.processOnePacket(DataGatherer.java:698)
	at uk.ac.manchester.spinnaker.front_end.download.DataGatherer$Downloader.doDownload(DataGatherer.java:661)
	at uk.ac.manchester.spinnaker.front_end.download.DataGatherer.fastDownload(DataGatherer.java:394)
	at uk.ac.manchester.spinnaker.front_end.download.DataGatherer.lambda$gather$2(DataGatherer.java:200)
	at uk.ac.manchester.spinnaker.front_end.download.DataGatherer$$Lambda$129/0x00000008401d4040.call(Unknown Source)
	at uk.ac.manchester.spinnaker.front_end.BasicExecutor.collectExceptions(BasicExecutor.java:77)
	at uk.ac.manchester.spinnaker.front_end.BasicExecutor.lambda$submitTasks$0(BasicExecutor.java:57)
	at uk.ac.manchester.spinnaker.front_end.BasicExecutor$$Lambda$130/0x00000008401d3840.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(java.base@11.0.11/FutureTask.java:264)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(java.base@11.0.11/ThreadPoolExecutor.java:1128)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(java.base@11.0.11/ThreadPoolExecutor.java:628)
	at java.lang.Thread.run(java.base@11.0.11/Thread.java:829)

   Locked ownable synchronizers:
	- <0x000000008f3a0bd8> (a java.util.concurrent.ThreadPoolExecutor$Worker)
```

dkfellows, 2022-04-26:
The lock itself doesn't seem to be the problem; thats in the guts of the selector and those are used in a thread safe way. More likely is a bad amount of time to wait and some good old fashioned missing events.

dkfellows, 2022-05-13:
Since it sounds (from other things you've said) like this is due to a chip being unreachable from its ethernet chip, I'm wondering why we're not detecting the timeout. Unless a core is crashing *after* sending the first message, when `received` would be true... Oh dear!

Perhaps a successful receive should reset the failure counter, and the number of failures before throwing `TimeoutException` should thus be smaller (because a success would turn the clock back).

rowleya, 2022-05-13:
I have done something like that now in the test code... more testing to be done!

PR #489 (2022-05-23): Rework the timeout logic for fast connections
The aim is that there's about 30 seconds (a bit over) after the last successful receive from a core before we time out the connection; any successful receive resets the counter. This lets use a shorter count of retries. (Previously, the timeout was effectively in the region of 200 seconds... if no messages were ever received, and infinite if one was received; a crashing core would make things go very badly wrong.)

I hope this fixes #453
commit 19ec9bd6 (2022-05-23): Rework the timeout logic for fast connections

The aim is that there's about 30 seconds (a bit over) after the last
successful receive from a core before we time out the connection; any
successful receive resets the counter. This lets use a shorter count of
retries.

I hope this fixes #453
closed 2022-05-23 by rowleya