SpiNNakerManchester/JavaSpiNNaker #500: Support updating of blacklist files?

rowleya, 2022-05-24:
A bit related to #429, I was wondering if it is also add an interface that would let admins blacklist cores and/or chips on the machine.  This should just require a BMP write of some sort, though some details would likely be required.

Some details how how this is currently done are here:
https://github.com/SpiNNakerManchester/SpiNNaker_hardware_tests/tree/master/blacklist_files

rowleya, 2022-05-24:
Safety features: 
 - Don't allow a user to blacklist the Ethernet chip of a board (or make that mean that they are blacklisting a board)
 - Don't allow a user to blacklist all cores on any chip (or make that mean that they are blacklisting a chip)

The above should stop a user blacklisting all the cores on an Ethernet chip too (as an edge case).

dkfellows, 2022-05-24:
Blacklist management would have to be an admin-only action.

rowleya, 2022-05-24:
> Blacklist management would have to be an admin-only action.

Absolutely!  But even admins make mistakes sometimes ;).  A sanity check might be good (just an error is fine; it doesn't have to change the meaning).

Christian-B, 2022-05-24:
I am 100% in favour of sanity checks even on Admins.  

For short admin tasks I favour clean and clear. NO because ...  rather than hidden I assume you mean ...

dkfellows, 2022-05-24:
Oh, this is "fun". The existing blacklists are stored using per-board coordinates (sensible because they're mostly worked out before the boards are commissioned in the machine) but those blacklists then use the (last part of the) serial number of the board to identify what they apply to. Which is fine... except nothing else knows what to do with that identifier.

Converting the blacklists we have into database entries is going to be "fun".

dkfellows, 2022-07-22:
There we go. No fancy GUI, but ought to work. Future changes can go on the general line.

PR #502 (2022-07-21): Manage hardware blacklists
fixes #500
closed 2022-07-22 by dkfellows