SpiNNakerManchester/JavaSpiNNaker #693: Undefined value for bare column sql group by

rowleya, 2023-01-19:
In some of the queries of the allocator a "group by" clause is used with "bare columns" (see https://www.sqlite.org/lang_select.html section 2.5).  This will lead to potentially incorrect results when e.g. selecting a rectangle starting at a specified coordinate or finding all possible rectangles including those with wrap arounds.

rowleya, 2023-01-19:
As an example, take a 12 board system.  It should be possible to query for a 2x1 triad rectangle starting at board with triad coordinates 1, 1, 0, even if that isn't really a valid allocation.

dkfellows, 2023-01-19:
Potential issue locations:
* `SQLQueries#GET_MACHINE_REPORTS`
  https://github.com/SpiNNakerManchester/JavaSpiNNaker/blob/550dfd865cf7b6187b89a3bc58f38b892c0e6717/SpiNNaker-allocserv/src/main/java/uk/ac/manchester/spinnaker/alloc/db/SQLQueries.java#L572
* queries/find_rectangle.sql
   https://github.com/SpiNNakerManchester/JavaSpiNNaker/blob/550dfd865cf7b6187b89a3bc58f38b892c0e6717/SpiNNaker-allocserv/src/main/resources/queries/find_rectangle.sql#L46
* queries/find_rectangle_at.sql
  https://github.com/SpiNNakerManchester/JavaSpiNNaker/blob/550dfd865cf7b6187b89a3bc58f38b892c0e6717/SpiNNaker-allocserv/src/main/resources/queries/find_rectangle_at.sql#L46
* queries/get_reported_boards.sql
  https://github.com/SpiNNakerManchester/JavaSpiNNaker/blob/550dfd865cf7b6187b89a3bc58f38b892c0e6717/SpiNNaker-allocserv/src/main/resources/queries/get_reported_boards.sql#L29

PR #694 (2023-01-19): Correct use of GROUP BY
This makes the code use `GROUP BY` and aggregates correctly, not relying on the DB to select a special row from the group for us. We simply can't be sure that the DB will choose correctly. Fortunately, we have indices already set up that let us efficiently handle this by using the coordinates of the root of the group (which we were grouping by) and inflating that back to the ID of the board at that point.

The report handling just looked wrong.

This fixes #693
commit 04102fa9 (2023-01-19): Correct use of GROUP BY

This makes the code use GROUP BY and aggregates correctly, not relying
on the DB to select a special row from the group for us. We simply can't
be sure that the DB will choose correctly. Fortunately, we have indices
already set up that let us efficiently handle this by using the
coordinates of the root of the group (which we were grouping by) and
inflating that back to the ID of the board at that point.

Fixes #693
closed 2023-01-19 by dkfellows