SpiNNakerManchester/JavaSpiNNaker #915: NullPointerException when using the admin jsp pages

rowleya, 2023-06-26:
When logged in as an administrator via jsp, clicking on the "Service Administration" link gives an error:
```
Whitelabel Error Page
This application has no explicit mapping for /error, so you are seeing this as a fallback.

Mon Jun 26 11:27:49 BST 2023
There was an unexpected error (type=Internal Server Error, status=500).
```

The server logs show the following:
```
2023-06-26 11:24:02.975  INFO 25211 --- [nio-8080-exec-4] u.a.m.s.a.s.LocalAuthProviderImpl        : authenticating Local Login UsernamePasswordAuthenticationToken [Principal=root, Credentials=[PROTECTED], Authenticated=false, Details=WebAuthenticationDetails [RemoteIpAddress=127.0.0.1, SessionId=365C13C99C42DF0F4DA3C58156F1F98F], Granted Authorities=[]]
2023-06-26 11:24:02.982  INFO 25211 --- [nio-8080-exec-4] u.a.m.s.a.s.LocalAuthProviderImpl        : User trust level is ADMIN
2023-06-26 11:24:05.239 ERROR 25211 --- [nio-8080-exec-7] o.a.c.c.C.[.[.[.[dispatcherServlet]      : Servlet.service() for servlet [dispatcherServlet] in context with path [] threw exception [Request processing failed; nested exception is java.lang.NullPointerException] with root cause

java.lang.NullPointerException: null
        at org.springframework.web.servlet.mvc.method.annotation.MvcUriComponentsBuilder.applyContributors(MvcUriComponentsBuilder.java:634) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.mvc.method.annotation.MvcUriComponentsBuilder.fromMethodInternal(MvcUriComponentsBuilder.java:552) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.mvc.method.annotation.MvcUriComponentsBuilder.fromMethodCall(MvcUriComponentsBuilder.java:320) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at uk.ac.manchester.spinnaker.alloc.web.ControllerUtils.uri(ControllerUtils.java:58) ~[classes!/:?]
        at uk.ac.manchester.spinnaker.alloc.admin.AdminControllerImpl.addStandardContextAttrs(AdminControllerImpl.java:195) ~[classes!/:?]
        at uk.ac.manchester.spinnaker.alloc.admin.AdminControllerImpl.addStandardContext(AdminControllerImpl.java:217) ~[classes!/:?]
        at uk.ac.manchester.spinnaker.alloc.admin.AdminControllerImpl.addStandardContext(AdminControllerImpl.java:233) ~[classes!/:?]
        at uk.ac.manchester.spinnaker.alloc.admin.AdminControllerImpl.mainUI(AdminControllerImpl.java:359) ~[classes!/:?]
        at uk.ac.manchester.spinnaker.alloc.admin.AdminControllerImpl$$FastClassBySpringCGLIB$$fdefff82.invoke(<generated>) ~[classes!/:?]
        at org.springframework.cglib.proxy.MethodProxy.invoke(MethodProxy.java:218) ~[spring-core-5.3.28.jar!/:5.3.28]
        at org.springframework.aop.framework.CglibAopProxy$CglibMethodInvocation.invokeJoinpoint(CglibAopProxy.java:793) ~[spring-aop-5.3.28.jar!/:5.3.28]
        at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:163) ~[spring-aop-5.3.28.jar!/:5.3.28]
        at org.springframework.aop.framework.CglibAopProxy$CglibMethodInvocation.proceed(CglibAopProxy.java:763) ~[spring-aop-5.3.28.jar!/:5.3.28]
        at org.springframework.security.access.intercept.aopalliance.MethodSecurityInterceptor.invoke(MethodSecurityInterceptor.java:67) ~[spring-security-core-5.8.4.jar!/:5.8.4]
        at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:186) ~[spring-aop-5.3.28.jar!/:5.3.28]
        at org.springframework.aop.framework.CglibAopProxy$CglibMethodInvocation.proceed(CglibAopProxy.java:763) ~[spring-aop-5.3.28.jar!/:5.3.28]
        at org.springframework.aop.framework.CglibAopProxy$DynamicAdvisedInterceptor.intercept(CglibAopProxy.java:708) ~[spring-aop-5.3.28.jar!/:5.3.28]
        at uk.ac.manchester.spinnaker.alloc.admin.AdminControllerImpl$$EnhancerBySpringCGLIB$$9bbc961a.mainUI(<generated>) ~[classes!/:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:205) ~[spring-web-5.3.28.jar!/:5.3.28]
        at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:150) ~[spring-web-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:117) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:895) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:808) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1072) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:965) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1006) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at org.springframework.web.servlet.FrameworkServlet.doGet(FrameworkServlet.java:898) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:529) ~[tomcat-embed-core-9.0.75.jar!/:?]
        at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883) ~[spring-webmvc-5.3.28.jar!/:5.3.28]
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:623) ~[tomcat-embed-core-9.0.75.jar!/:?]
        at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:209) ~[tomcat-embed-core-9.0.75.jar!/:?]
        at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:153) ~[tomcat-embed-core-9.0.75.jar!/:?]
        at org.apache.tomcat.websocket.server.WsFilter.doFilter(WsFilter.java:51) ~[tomcat-embed-websocket-9.0.75.jar!/:?]
        at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:178) ~[tomcat-embed-core-9.0.75.jar!/:?]
        at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:153) ~[tomcat-embed-core-9.0.75.jar!/:?]
        at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:111) ~[spring-web-5.3.28.jar!/:5.3.28]
        at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:178) ~[tomcat-embed-core-9.0.75.jar!/:?]
        at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:153) ~[tomcat-embed-core-9.0.75.jar!/:?]
        at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:337) ~[spring-security-web-5.7.8.jar!/:5.7.8]
        at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke(FilterSecurityInterceptor.java:115) ~[spring-security-web-5.7.8.jar!/:5.7.8]
        at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter(FilterSecurityInterceptor.java:81) ~[spring-security-web-5.7.8.jar!/:5.7.8]
        at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.8.jar!/:5.7.8]
        at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:122) ~[spring-security-web-5.7.8.jar!/:5.7.8]
        at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:116) ~[spring-security-web-5.7.8.jar!/:5.7.8]
        at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.8.jar!/:5.7.8]
[... cut ...]
        at org.apache.tomcat.util.net.SocketProcessorBase.run(SocketProcessorBase.java:52) [tomcat-embed-core-9.0.75.jar!/:?]
        at org.apache.tomcat.util.threads.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1191) [tomcat-embed-core-9.0.75.jar!/:?]
        at org.apache.tomcat.util.threads.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:659) [tomcat-embed-core-9.0.75.jar!/:?]
        at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61) [tomcat-embed-core-9.0.75.jar!/:?]
        at java.lang.Thread.run(Thread.java:829) [?:?]
```

dkfellows, 2023-06-26:
I *think* this is caused by those methods taking no arguments, which triggers some sort of edge case in the interceptor (which in turn is deep voodoo).

rowleya, 2023-06-26:
It is extra-odd since it used to work and now doesn't but from what I can see none of the code on the stack trace has changed between working and not working...

rowleya, 2023-06-26:
I am guessing this is caused by some other magic that has changed since therefore, such as a library version change somewhere... I suppose the most obvious thing would be whatever creates the ModelAndView instance that is returned by the controller methods in this case, which must have "null" for the arguments instead of an empty list perhaps...

commit f0c12fed (2023-06-26): Add dummy arguments to methods that don't really need them

Fixes #915
PR #916 (2023-06-26): Add dummy arguments to methods that don't really need them
Fixes #915
closed 2023-06-26 by rowleya