SpiNNakerManchester/JavaSpiNNaker #1028: Duplicate set before returning in EpochMap.removeSet

dkfellows, 2023-09-04:
I've been wondering whether this set should be copied so that the references become strong ones.

_Originally posted by @dkfellows in https://github.com/SpiNNakerManchester/JavaSpiNNaker/pull/992#discussion_r1314632101_

**NB:** Duplicate by using `WeakHashMap.forEach` to avoid ConcurrentModificationException, which means unwrapping things internally.

dkfellows, 2023-09-04:
The difficulty is, of course, that the wait set can be asynchronously modified and we can't stop it from happening or even really synchronize to delay it. Tricky stuff!

rowleya, 2023-09-04:
It is suggested here that the GC shouldn't cause a ConcurrentModificationException:
https://stackoverflow.com/questions/2861410/weakhashmap-iteration-and-garbage-collection

commit 052f6034 (2023-09-04): Make EpochMap safe against concurrent updates by the GC

This is very tricky code to fix #1028 properly. The hard part is
avoiding iterators, but it is necessary because those are not safe.
PR #1029 (2023-09-04): Make EpochMap safe against concurrent updates by the GC
This is very tricky code that fixes #1028 properly. The hard part is avoiding iterators, but it is necessary because those are not safe.
closed 2023-09-04 by dkfellows