SpiNNakerManchester/JavaSpiNNaker #1036: If power requests are in progress errors can occur on some requests

rowleya, 2023-09-07:
This is a specific issue that can happen only under certain specific circumstances.  An example is:
1) User creates new Job request for a board that is currently allocated.
2) The request is queued.
3) The allocation holding the board finishes and starts the process of powering off.
4) At the same time (or shortly after - but before the power request is picked up) the user cancels the job.
5) The user cancelling the job will see an error similar to the following:
```
spalloc_client.protocol_client.SpallocServerException: PreparedStatementCallback; SQL [-- Copyright (c) 2021 The University of Manchester
       --
       -- Licensed under the Apache License, Version 2.0 (the "License");
       -- you may not use this file except in compliance with the License.
       -- You may obtain a copy of the License at
       --
       --     https://www.apache.org/licenses/LICENSE-2.0
       --
       -- Unless required by applicable law or agreed to in writing, software
       -- distributed under the License is distributed on an "AS IS" BASIS,
       -- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
       -- See the License for the specific language governing permissions and
       -- limitations under the License.
       
       INSERT INTO pending_changes(
           job_id, board_id, from_state, to_state,
           power, fpga_n, fpga_e, fpga_se, fpga_s, fpga_w, fpga_nw)
           VALUES (
               	?, ?, ?, ?,
               	?, ?, ?, ?, ?, ?, ?);
           ]; 
Duplicate entry '9161' for key 'pending_changes.board_id'; nested exception is java.sql.SQLIntegrityConstraintViolationException: Duplicate entry '9161' for key 'pending_changes.board_id'
```

We could simply add an "ON DUPLICATE KEY..." to the statement, but it might need more logic e.g. the from_state and to_state could be different for example.

dkfellows, 2023-09-11:
Maybe a `WHERE NOT EXISTS` clause would be more helpful?

rowleya, 2023-09-11:
That is similar to the ON DUPLICATE KEY solution.  The problem is that the job is in the middle of transitioning between states, then a user asks to transition to what could be a different state e.g. it is currently powering on but the user decides to destroy it.  This may need additional handling in the Java code rather than the SQL code...

rowleya, 2023-09-15:
It looks like this was probably a left-over constraint on the main database; having removed the table in question and let it be recreated, it seems to now let me get away with this!  I will leave this until the next Integration testing goes through to see if it comes up again...

closed 2025-01-10 by rowleya