SpiNNakerManchester/JavaSpiNNaker #1297: Add support for Google Authentication

rowleya, 2025-09-01:
Allow Google authentication here, which can then be used by our local Jupyter server.  This may need a bit more thought since there are certain things in the OpenID login that rely on EBRAINS things, like Collabs and Quota.

Christian-B, 2025-09-01:
can google users not use private clooabs as setup in https://github.com/SpiNNakerManchester/JavaSpiNNaker/pull/1296

Christian-B, 2025-09-01:
The whole Collabs and Quota stuff has been a lot less useful than hope so this may even be the motivation for stripping that completely out.

rowleya, 2025-09-01:
We do use the quota and collabs, but only from EBRAINS.  A Google user would probably have no specific quota; this would be left to EBRAINS for now.

closed 2026-11-13 by rowleya